hawku/TabletDriver #670: Trying to lower hovering distance - Please help

MrBigNoodle, 2020-06-03:
Hi I'm trying to use my Gaomon S620 in Fortnite in mouse mode. Everything works fine except that the tablet detects the pen when it's a solid inch above the tablet. As an example if I slide my pen across the tablet all the way to the right and when I reach the edge of the tablet (or tablet area) I want to move the pen back to the center but without moving the cursor/crosshair, and without having to lift the pen like 2 inches off the tablet each time. I just want to lift the pen a tiny bit. In fact if it's possible I'd even disable the ability for the tablet to detect the pen at all unless it's actually touching the tablet. But at the minimum I'd like to reduce the distance. Can someone PLEASE help me? 
It's sometimes called PressureSafteyMargin apparently, in other programs. 

## Hardware/software information
Windows Version: 10

Tablet Driver version: v0.3.1a

Tablet model: Gaomon S620

MrBigNoodle, 2020-06-03:
I FOUND THE ANSWER. Mods can delete this if they want but if anyone else is looking for the answer, you can put "RelativeDragMove true" (Without quotes) in the command section and it works. Apparently you can combine that with Clearing the tip button mapping from the buttons tab to do other things, but I didn't need to do that.

MrBigNoodle, 2020-06-03:
Someone should add this as a feature. Would be really easy to implement since it's just a command.

elloot, 2020-12-27:
@MrBigNoodle I know this issue is quite old by now but I just ran into the same problem as you. I also want to use my tablet for shooter games with this driver but can't because of the hover distance. The fix you suggested doesn't work for me, I have tried using the command you posted but apparently that doesn't exist anymore. When I run `RelativeDragMove true` I get a warning as a response telling me that `RelativeDragMove` is an unknown command. Have you also ran into this issue and if so how did you fix it?