hawku/TabletDriver #671: Tablet not detected by driver. (G640)

MrWawas, 2020-06-03:
I've uninstalled and reinstalled the OEM drivers, and this driver 10+ times at this point, to no avail.
It always comes up with the same message, "Driver stopped. Restarting! Check console !!!" with the console saying "Tablet not found! Check the list of supported tablets."

I've also tried uninstalling the drivers, and device, from Device Manager. All with restarting my computer between installs/uninstalls.

Any help is appreciated, thanks in advance! :)

Start up log attached below;

[startuplog.txt](https://github.com/hawku/TabletDriver/files/4725685/startuplog.txt)

## Hardware/software information
Windows Version: Windows 10, 1909

Tablet Driver version: 0.2.3

Tablet model: XP-Pen Star G640

ILikeNoodles0, 2020-06-04:
I have the same problem and same console message :(

Baha290, 2020-06-05:
@MrWawas I have the same problem

WonderfulDay0, 2020-06-06:
same problem

ssccoonniicc, 2020-06-07:
You may have the second revision of the G640.
Check: https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/

Baha290, 2020-06-07:
@ssccoonniicc it worked thank you.

MrWawas, 2020-06-08:
> You may have the second revision of the G640.
> Check: https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/

For anybody else having the issue; make sure to check @ssccoonniicc's link and try that config.

closed 2020-06-08 by MrWawas