hawku/TabletDriver #676: Tablet Drivers doesnt open

NorthGalaxy, 2020-06-06:
It worked but it doesnt anymore, the exe file opens but instantly says  "TabletDriverGUI has stopped working"

## Hardware/software information
Windows Version: Windows 7 Ultimate

Tablet Driver version: 0.2.3

Tablet model: wacom one ctl 672

NorthGalaxy, 2020-06-06:
it worked on windows 10*

Vlothx, 2020-06-11:
Install all updates available for your system. Common problem with Windows 7, that's why I don't recommend it anymore.

NorthGalaxy, 2020-06-13:
> Install all updates available for your system. Common problem with Windows 7, that's why I don't recommend it anymore.

any specific updates to install?

Vlothx, 2020-06-15:
All of them, should be like 6GB of updates (if you haven't updated your system).

NorthGalaxy, 2020-06-20:
Fixed it by install vcredist 86 bit

closed 2020-06-20 by NorthGalaxy