hawku/TabletDriver #679: Choppy Inputs on Windows 10 2004 Update

MaxKruse, 2020-06-08:
After installing the newest windows update, the input of tablets will be extremely laggy and smoothened.

## Hardware/software information
Windows Version: 19041.264

Tablet Driver version:  0.1.5.5 Devocub

Tablet model: CTL 470(k)

ssccoonniicc, 2020-06-08:
Smoothing is pretty broken on win 10 2004 but somebody has fixed it: https://github.com/AbdShullah/TabletDriver

MaxKruse, 2020-06-08:
good to know, thanks.

Vlothx, 2020-06-11:
Sadly, no. Microsoft changed the behaviour of the CreateTimerQueueTimer WinApi function call. The only way of getting this working is using another high resolution timer, and that's what Abd Shullah's did with his fork. High resolution timers are quite resource intensive.

Pizzabelly, 2020-06-24:
@MaxKruse can you try this version? https://github.com/Pizzabelly/TabletDriver/releases/tag/v0.1.5.5

MaxKruse, 2020-06-25:
> @MaxKruse can you try this version? https://github.com/Pizzabelly/TabletDriver/releases/tag/v0.1.5.5

very good fix, all works normal!

closed 2020-06-25 by MaxKruse