hawku/TabletDriver #681: drivers not working with XP pen 6430s

awesome1223, 2020-06-11:
basically, I've been using the g430s for a while now. I needed a new one due to scratches on my old tablet, I ordered a new one (the exact same tablet), but for some reason it wont work with the drivers. It doesn't pick up the tablet,,, but it still picks up my old one. I tried all the fixes I could find on here/reddit but nothing has worked so far,

## Hardware/software information
Windows Version: 1803

Tablet Driver version: 0.2.3

Tablet model: XP-Pen g430s
[startuplog.txt](https://github.com/hawku/TabletDriver/files/4762006/startuplog.txt)

Vlothx, 2020-06-11:
That's because, you might have a new revision. You can try with InfinityGhost's fork of TabletDriver (link [here](https://github.com/InfinityGhost/TabletDriver)). If you can't get it work, you can join the OpenTabletDriver discord and we might be able to help you with your problem.

awesome1223, 2020-06-11:
holy shit it worked, thank you so much!!!!

closed 2020-06-11 by awesome1223