hawku/TabletDriver #684: New tablets xp-pen g640 doesn't with driver. Please fix :3

Yatagarasu91, 2020-06-13:
## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Yatagarasu91, 2020-06-13:
Doesn't work with xp-pen g640. Maybe it's just a new version

Vlothx, 2020-06-15:
[Look at this r/osugame post.](https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/)

Yatagarasu91, 2020-06-15:
I'll try, if that works I will say about it. Thanks for help

Yatagarasu91, 2020-06-15:
I still can't install vmulti driver "devcon.exe failed"

jorji3, 2020-06-17:
your title disgusts me but sadly i am having the same issue with my XP Pen tablet after a haitus from osu. i'd really like some input on the issue :| 

```
  Are you sure you want to install vmulti driver?

Press enter to install...

C:\Users\Joji\Desktop\TabletDriverV0.2.2>driver_vmulti\install_vmulti_64.bat

C:\Users\Joji\Desktop\TabletDriverV0.2.2>REM  --> Check for permissions

C:\Users\Joji\Desktop\TabletDriverV0.2.2>"C:\WINDOWS\system32\cacls.exe" "C:\WINDOWS\system32\config\system"   1>nul 2>&1

C:\Users\Joji\Desktop\TabletDriverV0.2.2>REM --> If error flag set, we do not have admin.

C:\Users\Joji\Desktop\TabletDriverV0.2.2>if '0' NEQ '0' (
echo Requesting administrative privileges...
 goto UACPrompt
)  else (goto gotAdmin  )

C:\Users\Joji\Desktop\TabletDriverV0.2.2>if exist "C:\Users\Joji\AppData\Local\Temp\getadmin.vbs" (del "C:\Users\Joji\AppData\Local\Temp\getadmin.vbs"  )

C:\Users\Joji\Desktop\TabletDriverV0.2.2>pushd "C:\Users\Joji\Desktop\TabletDriverV0.2.2"

C:\Users\Joji\Desktop\TabletDriverV0.2.2>CD /D "C:\Users\Joji\Desktop\TabletDriverV0.2.2\driver_vmulti\"

C:\Users\Joji\Desktop\TabletDriverV0.2.2\driver_vmulti>devcon.exe /r install vmulti.inf pentablet\hid
devcon.exe failed.
Press enter...```

attempting to use version 0.22 as v.23 does the same thing for me.

Vlothx, 2020-06-17:
Try uninstalling all VMulti drivers that you have installed, and then, try to install the 32 bit version of VMulti.

jorji3, 2020-06-17:
Okay, well that mostly worked but I'm starting to suspect that it was a Windows Batch api error, as windows prompted me that it needed to install something, and I followed suit. After that I ran the 32bit version of vmulti_install instead and it worked. Maybe OP can try this out as well, no clue. But it is fixed now!

Vlothx, 2020-06-17:
You had the "devcon.exe failed." error because most probably you were trying to install a 64 bit driver into a 32 bit os.

jorji3, 2020-06-17:
my os is most definitely 64bit. Im clueless as to what was the cause of the issue.
[image]

Yatagarasu91, 2020-06-18:
i have 64 bit oc but vmulti driver can't be installed cuz "devcon.exe failed"

Yatagarasu91, 2020-06-18:
> your title disgusts me but sadly i am having the same issue with my XP Pen tablet after a haitus from osu. i'd really like some input on the issue :|
> 
> ```
>   Are you sure you want to install vmulti driver?
> 
> Press enter to install...
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>driver_vmulti\install_vmulti_64.bat
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>REM  --> Check for permissions
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>"C:\WINDOWS\system32\cacls.exe" "C:\WINDOWS\system32\config\system"   1>nul 2>&1
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>REM --> If error flag set, we do not have admin.
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>if '0' NEQ '0' (
> echo Requesting administrative privileges...
>  goto UACPrompt
> )  else (goto gotAdmin  )
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>if exist "C:\Users\Joji\AppData\Local\Temp\getadmin.vbs" (del "C:\Users\Joji\AppData\Local\Temp\getadmin.vbs"  )
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>pushd "C:\Users\Joji\Desktop\TabletDriverV0.2.2"
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2>CD /D "C:\Users\Joji\Desktop\TabletDriverV0.2.2\driver_vmulti\"
> 
> C:\Users\Joji\Desktop\TabletDriverV0.2.2\driver_vmulti>devcon.exe /r install vmulti.inf pentablet\hid
> devcon.exe failed.
> Press enter...
> ```
> 
> attempting to use version 0.22 as v.23 does the same thing for me.
> 
> small edit: running it as admin just makes the CMD prompt disappear with no confirmation message, then opening TabletDriver just causes me to be met with an unresponsive but "Connected" tablet. Xp-pen G430s.
> 
> [image]
> 
> seemingly "working" but no response from my tablet whatsoever. default xp pen drivers worked on this same pc, hawku drivers just refuse to communicate past finding a device signature?

xp-pen few weeks ago maybe month ago released new version of G640 with a little changes, i thought that is a problem and that's why driver won't work