hawku/TabletDriver #688: PC doesn't start

alsoalex, 2020-06-16:
I installed the drivers and my pc automatically restarted, and then windous is loading for so many minutes.. idk why tho
[image]

## Hardware/software information
Windows Version: 10 pro

Tablet Driver version: 

Tablet model: xp pen star g640s

alsoalex, 2020-06-16:
still starting...

alsoalex, 2020-06-16:
now it's just black screen.....

alsoalex, 2020-06-16:
i reinstalled my windous... thanks..

ssccoonniicc, 2020-06-16:
I've seen a few other people say that windows was slower/wouldn't start after vmulti but I never found a solution to that. If you want you can try [opentabletdriver](github.com/InfinityGhost/OpenTabletDriver) since it doesn't use vmulti.