hawku/TabletDriver #693: need help

JetFury, 2020-06-19:
i opened the driver bcus i was going to play osu, after that when i was moving my cursor with the pen, the report rate was about 60-70(it shows that amount when I move the cursor in osu), so the report rate is very low, but in the driver im using 500 report/polling rate, why is that?

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: g640s xp pen

ssccoonniicc, 2020-06-20:
if you are on windows 10 2004 then smoothing is kinda broken but it is fixed here: https://github.com/AbdShullah/TabletDriver though it uses a resource intensive timer

JetFury, 2020-06-20:
oof tablet not supported

ssccoonniicc, 2020-06-20:
just copy the tablet config file from your current driver and put it in the other driver

ssccoonniicc, 2020-06-20:
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00
```