hawku/TabletDriver #695: Tablet not found (XP-Pen G640)

NotIPlayForFun, 2020-06-22:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G640
[image]

NotIPlayForFun, 2020-06-22:
Ok so I just realized that this is apparently because of a revision of the tablet. Still, can anyone tell me how bad this is? I just got the tablet and I wonder if I should refund it? Can I even still get the old version?? Any help would be much appreciated :P

ssccoonniicc, 2020-06-22:
the second revision has 1/2 the resolution of the first tablet, a lower polling rate (by about 66z, 266hz-200hz) and more hardware smoothing. You can read more [here](https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/). You can most likely get a refund but unless you find the first revision used you probably wont be able to find it. You should probably get the Gaomon s620 instead and use InfinityGhost's TabletDriver fork (because hawku's driver doesn't have g620 support.)

NotIPlayForFun, 2020-06-22:
> the second revision has 1/2 the resolution of the first tablet, a lower polling rate (by about 66z, 266hz-200hz) and more hardware smoothing. You can read more [here](https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/). You can most likely get a refund but unless you find the first revision used you probably wont be able to get it. You should probably get the Gaomon s620 instead and use InfinityGhost's TabletDriver fork (because hawku's driver doesn't have g620 support.)

Thanks! May I ask: What do you think about the Wacom 480 aka Intuos S for 65€? It seems like a good option :P Just in case maybe you have some experience/opinion or whatever on that, since as this is my first tablet I don’t rly know whats the good options out there, and reviews are pretty much always paid :D

ssccoonniicc, 2020-06-24:
65 Euros is an ok price for a 480 but you could also get a Wacom 472 or 471 because they have the same digitizer as the 480. The 480 is a really good tablet though and I would recommend it but you should definitely consider a 472 or 471. I was able to get my 480 on Facebook marketplace for $30(~27€).