hawku/TabletDriver #697: i have a g640 and i can't get it to work pls help

JustinMiroux, 2020-06-24:
## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Fogazzle, 2020-06-24:
I'm at work rn but I have the solution it's in a comment section of one of the youtube videos about it. Add me on discord and I will send it to you later. Fogazzle#0498

Fogazzle, 2020-06-24:
[image]

Fogazzle, 2020-06-24:
Follow this exactly. It worked for me

JustinMiroux, 2020-06-24:
i can't edit the file

Fogazzle, 2020-06-24:
I'm about to get off I'll help you when I get home

bibeoo, 2020-06-28:
bro can you paste the text here ?

Fogazzle, 2020-06-30:
Hopefully this fixes it just make sure you go into the "Config" file next open the "tablet" file with preferably Note Pad scroll all the way down towards the bottom to be safe and tap the enter key  twice after doing so paste this into the bottom of the note pad then exit and save.

XP-Pen G640
HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

Fogazzle, 2020-06-30:
@bibeoo