hawku/TabletDriver #699: (Solution) Xp pen g640 driver stopped. Restarting.

Fogazzle, 2020-06-24:
This is a fix I found on youtube from a comment section.
[image]

## Hardware/software information
Windows Version: 

Tablet Driver version: hakwu

Tablet model: g640

Fogazzle, 2020-06-24:
Hope this helps

Vlothx, 2020-06-26:
Yeah, second revision of the G640. [Here's] (https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/) an r/osugame post a about that.

Fogazzle, 2020-06-30:
Hopefully this fixes it just make sure you go into the "Config" file next open the "tablet" file with preferably Note Pad scroll all the way down towards the bottom to be safe and tap the enter key  twice after doing so paste this into the bottom of the note pad then exit and save.

XP-Pen G640
HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00