hawku/TabletDriver #702: g640 debug problems

loeizbi, 2020-06-24:
Could someone help me solve theses i don't understand what i need to do :(

2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: 'Kingston' Product: 'HyperX Virtual Surround Sound'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x0951, Product Id: 0x16A4
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0x000C, Usage: 0x0001
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 33, OutputLen: 37
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: '' Product: 'EP2 Interrupt'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x04B4, Product Id: 0x0101
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0x0001, Usage: 0x0080
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 2, OutputLen: 0
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: '' Product: 'EP2 Interrupt'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x04B4, Product Id: 0x0101
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0x000C, Usage: 0x0001
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 2, OutputLen: 0
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0001
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 65, OutputLen: 65
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: 'Kingston' Product: 'HyperX Virtual Surround Sound'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x0951, Product Id: 0x16A4
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0x000B, Usage: 0x0005
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 2, OutputLen: 2
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0002
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 33, OutputLen: 65
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0x0000, Usage: 0x000E
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 3, InputLen: 0, OutputLen: 0
2020-06-24 23:39:46 [HIDDevice] [DEBUG]
2020-06-24 23:39:46 [HIDDevice] [DEBUG] HID Device: Vendor: 'UGTABLET' Product: '6 inch PenTablet'
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Vendor Id: 0x28BD, Product Id: 0x0914
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   Usage Page: 0xFF0A, Usage: 0x0001
2020-06-24 23:39:46 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 12, OutputLen: 12
2020-06-24 23:39:46 [HIDDevice] [DEBUG]

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Vlothx, 2020-06-26:
You have the second revision of the G640, look at this for more: httpss://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/

loeizbi, 2020-06-26:
Thanks for your time vlothx