hawku/TabletDriver #705: Built TabletDriverGUI.exe on latest master is being detected as a false positive virus.

X9VoiD, 2020-06-29:
VirusTotal: [https://www.virustotal.com/gui/file/1a90c050d416088ebc174d297a4c1e3aa0d25e077623fd5073265b699eb9e767/detection](url)

My guess is that we are being detected as a false positive due to the new changes on screen map. TabletDriverGUI.exe captures the screen which might be triggering the false positive.

X9VoiD, 2020-06-29:
Should I git bisect to confirm?

X9VoiD, 2020-06-29:
So, I git bisected anyway, and it showed dbcbb5844a947cbee6fd3745039e7292f2aae6d8 as the first bad commit.

For reference, here is v0.2.3 's VirusTotal [result](https://www.virustotal.com/gui/file/e9068b53428d189acb242d108b0f4896e43640a88e51e3aa1880677017c383c4/detection)
And here is dbcbb5844a947cbee6fd3745039e7292f2aae6d8's VirusTotal [result](https://www.virustotal.com/gui/file/ca2a8e4f258d2ba15db2df85657cc6511e6e691f7bf760040b474b59a733a1b9/detection)

InfinityGhost, 2020-06-29:
This problem is the reason I stopped making additions to my fork, I think that what you said is correct however. I believe that the screen capture method that commit added is used in some sort of malware that's triggering the false positive.