hawku/TabletDriver #709: Not supported tablet G640S

Michaelbabidi, 2020-07-01:
Hi so i was wondering if it was possible to take the g640s settings from the hawku driver to your devocub driver so the g640s can work on your drivers?

## Hardware/software information 
Windows Version: 10

Tablet Driver version: G640S

Tablet model: G640S

Vlothx, 2020-07-01:
Should be possible, you have to copy the G640S entry (from tablet.cfg) and paste it to the tablet.cfg of the DevoCub fork.

Michaelbabidi, 2020-07-01:
I tried pasting in the g640s entry into devocubs config and it still doesnt recognize my tablet

Vlothx, 2020-07-01:
Try this config:

`#
# XP-Pen G640s
#
Tablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00`

Michaelbabidi, 2020-07-01:
Yes that worked thank you very much! The two entries are very similar but in the hawku one which i copied from it had HIDTablet
And in yours it was just Tablet which weirdly enough makes the difference

closed 2020-07-01 by Michaelbabidi