hawku/TabletDriver #710: Uninstalling

Kolba1t, 2020-07-03:
it's labeled as a trojan virus which I know is an issue people have had, idk if people have had windows defender uninstall it randomly though, so I'm reporting it just in case. Windows defender will just randomly uninstall the driver entirely after I've allowed it on my pc, is there any way to fix this? it's really frustrating and even if I turn off windows defender it turns itself back on because microsoft is fucking stupid. 

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.3.1

Tablet model: goamon s620

Vlothx, 2020-07-03:
Add an exclusion.

Kolba1t, 2020-07-03:
> Add an exclusion.

how do I do that? what does that mean? sorry I don't know much about computer stuff or coding

Vlothx, 2020-07-03:
If you're using Microsoft Defender, [here](https://support.microsoft.com/en-us/help/4028485/windows-10-add-an-exclusion-to-windows-security)'s a guide from Microsoft itself. You have to add the entire folder to the exclusion list.