hawku/TabletDriver #711: Tablet Driver not working

Espioo, 2020-07-04:
Hello,

I recently uninstalled every Wacom driver for my tablet in order to change to the Tablet Driver (I had a hard time but I think I did it right), and now that I have Tablet Driver, I can't have the tablet area I want, it only uses full-size area, even though I unchecked the box and I changed to my old area.

[image]

Here's a screenshot of my driver settings.

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: Bamboo CTH-470/S

Vlothx, 2020-07-04:
As you can see from the window title:

> Wacom drivers installed

What you have to do here is, install the Wacom driver again -> uncheck the Windows Ink box (Wacom Tablet Properties) -> uninstall the driver.

Espioo, 2020-07-05:
As easy as that, it works perfectly now, thanks! <3

Now you obviously must change your profile bio