hawku/TabletDriver #714: Tablet as joystick, new meta mouse-controller alternative

AuracleTech, 2020-07-06:
I'd like to make a function to game on FPS games with the tablet, like mouse, but if you come back towards the center, the mouse doesn't get dragged, like a joystick. Kind of hard to explain, I never saw this anywhere.

I found out that playing with tablet on FPS can give you super accuracy, but the fact that I have to leave the pen and go back to the center to reset position is bad.

I'd need your discord and/or contact so I can work with you on this!
I can't label this as an enhancement more than an issue 😁

AuracleTech, 2020-07-11:
[image]
Also I made myself a skin, didn't like the original

Vlothx, 2020-07-11:
Welp, I can't understand how you want it to work. But uh... if I remember correctly there was a command in InfinityGhost's fork of TabletDriver so the cursor moves only when pressure is applied, might help with what you want to do?

AuracleTech, 2020-07-13:
Not what I am trying to do :/
Hard to explain sadly... watch a video of someone gaming and imagine controlling using a tablet without leaving the tablet area or lifting the pen, this is what I am trying to do.

AuracleTech, 2022-03-16:
I will close this issue as I don't know how to explain it further

closed 2022-03-16 by AuracleTech