hawku/TabletDriver #716: TabletDriver doesnt see MSVCP140.dll and VCRUNTIME140.dll

CherryBloom1, 2020-07-07:
## Hardware/software information
Windows Version: 10 / x64/ build - 2004

Tablet Driver version: 0.2.3

Tablet model: XP-PEN G640S

## Other

Video - https://youtu.be/jJDeQdixQjY

rconjoe, 2020-07-16:
Having the same issue. Have already applied the G640 config fix. This worked previously with no issues...

CherryBloom1, 2020-07-16:
> Having the same issue. Have already applied the G640 config fix. This worked previously with no issues...
> 
> EDIT: Explains why. I reinstalled windows a few times this week (totally unrelated) and forgot to install:
> 
> * https://www.microsoft.com/net/download/dotnet-framework-runtime
> * https://aka.ms/vs/15/release/vc_redist.x86.exe
> 
> Fixed everything for me.

CherryBloom1, 2020-07-16:
thanks, rconjoe!

closed 2020-07-16 by CherryBloom1

kreksoverzum, 2020-08-26:
I want to mention it explicitly, as for me, error was solved after installing x86 redist although i am on x64 system. Installing x64 redist changed nothing.