hawku/TabletDriver #717: The Driver doesn't work.

aimbotcone, 2020-07-07:
The driver is not working. Starting it as Administrator doesn't work reininstall The Tabletdriver and the VMulti driver doesn't work aswell, I have uninstalled everything with XP-PEN and still doesn't work.

When i want to download the VMulti driver on this site, i cant the link for XP-PEN doesn't work.

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Vlothx, 2020-07-07:
You have the second revision of the G640, [look at this](https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/).

aimbotcone, 2020-07-07:
> You have the second revision of the G640, [look at this](https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/).

Where can i find the config folder?

aimbotcone, 2020-07-07:
> You have the second revision of the G640, [look at this](https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/).

Doesn't fix it. It says the same problem as before, nothing changed.

aimbotcone, 2020-07-08:
Okay i fixxed it, i just installed the XP-PEN Driver again so that the tablet works basically and it fixxed the 2:1 problem in Hawku too.  After that i uninstalled Pentablet the xp-pen driver and restarted my PC, idk how i fixxed it but that worked. Still thanks for the help! :)