hawku/TabletDriver #721: it keeps on saying "can't open XP-Pen's VMulti" can someone pls help?

NirDragneel, 2020-07-09:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: XP-Pen StarG640
[image]

AdCb0, 2020-07-09:
Hi. Just run "install_vmulti_driver.bat"

[image]

NirDragneel, 2020-07-09:
I did. I ran it and removed it and ran it again and also restarted my computer nothing seems to work.

Vlothx, 2020-07-11:
Try InfinityGhost's fork of TabletDriver.

NirDragneel, 2020-07-12:
> Try InfinityGhost's fork of TabletDriver.

I just downloaded it and windows defender deleted it as a virus instantly....

Vlothx, 2020-07-12:
That's a false positive, you can disable your antivirus and add an exclusion to the entire folder.