hawku/TabletDriver #723: Can't open 'init.cfg'

godgamergang, 2020-07-11:
The console on the driver said that it can't start the driver because bin/TabletDriverService.exe not found! 
I went to go and download it but it said it couldn't open 'init.cfg'

Vlothx, 2020-07-11:
Please extract TabletDriver to a folder, don't run it directly from the .zip

godgamergang, 2020-07-12:
where is TabletDriver

godgamergang, 2020-07-12:
or is that the entire driver setup because i did extract it to a folder

Vlothx, 2020-07-12:
Try InfinityGhost's fork of TabletDriver, you might have better luck with that. Don't forget to disable your antivirus so you can download it, and also add an exclusion so your antivirus doesn't delete the driver.

godgamergang, 2020-07-13:
can i have a link to it

Vlothx, 2020-07-13:
https://github.com/InfinityGhost/TabletDriver