hawku/TabletDriver #734: Driver Stopped-Tablet not foud Xp-Pen g430S

Sosa777, 2020-07-18:
Hello, i've followed all the Steps on the main Guide to install Hawku Drivers, i bought the tablet and then i've installed the Hawku Drivers for first, istead of using the original XP pen drivers, so i should not unistall nothing. But the GUI still stopped the driver. I can't understand why or where i can found a Fix guide, i searched a few guides but no one worked for me.

[startuplog.txt](https://github.com/hawku/TabletDriver/files/4941911/startuplog.txt)

Vlothx, 2020-07-18:
You'll have to use InfinityGhost's fork of TabletDriver for that, [here's the link](https://github.com/InfinityGhost/TabletDriver).

Sosa777, 2020-07-18:
> You'll have to use InfinityGhost's fork of TabletDriver for that, [here's the link](https://github.com/InfinityGhost/TabletDriver).

Mate u're really my savior!! It's fixed <3

closed 2020-07-18 by Sosa777