hawku/TabletDriver #739: Running "install_vmulti_driver.bat" and not having the TabletDriverGUI.exe file

Jenno1, 2020-07-23:
Hello. I was trying to download this and followed the instructions until it told me to run "install_vmulti_driver.bat" which i did and got the error message "windows can't find install_vmulti_driver.bat, and so on" I also looked through everything i downloaded and didn't find the TabletDriverGUI.exe file. Please help me. Thank you!

## Hardware/software information
Windows Version: win 10 64bit

Tablet Driver version: haven't even downloaded it :/

Tablet model: veikk s640

Vlothx, 2020-07-23:
Have you tried, you know...? Downloading it? http://hwk.fi/TabletDriver/TabletDriverV0.2.3.zip

Jenno1, 2020-07-23:
> Have you tried, you know...? Downloading it? http://hwk.fi/TabletDriver/TabletDriverV0.2.3.zip

Yes...

Jenno1, 2020-07-23:
> Have you tried, you know...? Downloading it? http://hwk.fi/TabletDriver/TabletDriverV0.2.3.zip

nvm im just fucking stupid

closed 2020-07-23 by Jenno1