hawku/TabletDriver #746: Tablet G640 / Check Tablet , Tablet Not Found

NeoBurgerYT, 2020-07-29:
## Hardware/software information
Windows Version: Window 10 64 bits

Tablet Driver version: 0.1.5

Tablet model: Xp pen g640

2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: 'UGTABLET' Product: '6 inch PenTablet'
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x28BD, Product Id: 0x0914
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0xFF0A, Usage: 0x0001
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 12, OutputLen: 12
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: '6 inch PenTablet' Product: '6 inch PenTablet'
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x1B1C, Product Id: 0x1BF0
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0x000C, Usage: 0x0001
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 3, OutputLen: 0
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: '' Product: ''
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x1B1C, Product Id: 0x1B15
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0xFFC2, Usage: 0x0004
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 65, InputLen: 65, OutputLen: 65
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: '' Product: ''
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x1B1C, Product Id: 0x1B15
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0x000C, Usage: 0x0001
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 5, OutputLen: 0
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: '' Product: ''
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x1B1C, Product Id: 0x1B15
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0xFFC0, Usage: 0x0002
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 64, OutputLen: 0
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0x0000, Usage: 0x000E
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 3, InputLen: 0, OutputLen: 0
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0001
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 65, OutputLen: 65
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0002
2020-07-29 06:15:20 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 33, OutputLen: 65
2020-07-29 06:15:20 [HIDDevice] [DEBUG] 
2020-07-29 06:15:20 [INFO] 
2020-07-29 06:15:20 [INFO]   Driver version: 0.1.5
2020-07-29 06:15:20 [INFO]   Windows version: Microsoft Windows NT 6.2.9200.0
2020-07-29 06:15:20 [INFO]   Windows product: Windows 10 Home
2020-07-29 06:15:20 [INFO]   Windows release: 1903
2020-07-29 06:15:20 [INFO] 
2020-07-29 06:15:20 [INFO] >> CheckTablet
2020-07-29 06:15:20 [ERROR] Tablet not found!
2020-07-29 06:15:20 [ERROR] Check the list of supported tablets from the GitHub page.
2020-07-29 06:15:20 [ERROR] http://github.com/hawku/TabletDriver

Hello , my graphics tablet is not recognized by the driver

DJDuckBerlin, 2020-07-29:
I recently had the same problem. There is a newer model of the G640. Therefore the tablet is not recognized by the program. They've  changed some things. To fix it, you need to go into the drivers folder and open the "tablet.cfg" file with an editor porgram.

You need to change the properties to this:
HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
#CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

Also make sure to update the drivers to the newest version and uninstill all other tablet drivers on your system.
I hope its gonna work ; )

[image]

theArcticGiant, 2020-07-31:
I'm having the same issue, but it's a XP-PEN G430S that I bought to replace my old tablet. It arrived today, so very new.

NeoBurgerYT, 2020-08-04:
> I recently had the same problem. There is a newer model of the G640. Therefore the tablet is not recognized by the program. They've changed some things. To fix it, you need to go into the drivers [...]

i can't use an older plugin ?