hawku/TabletDriver #749: ERROR! Can't start the driver! bin/TabletDriverService.exe not found!

chowman3, 2020-08-02:
Windows Version: 10
Tablet Driver version: 0.2.3
Tablet model: xp-pen star g640
[image]

Vlothx, 2020-08-02:
Extract the driver to a folder, don't run it directly from the .zip

chowman3, 2020-08-02:
just extracted it to my desktop and it didn't work. would i need to extract it somewhere specifically?

chowman3, 2020-08-02:
nvm now it says tablet not found

chowman3, 2020-08-02:
[image]

ssccoonniicc, 2020-08-02:
When did you buy your tablet? If you bought it recently you need to add
`#
# XP-Pen G640_B
#
HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
CheckString Product "6 inch PenTablet"
Name "XP-Pen G640_B"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00`
to tablet.cfg

chowman3, 2020-08-03:
bought it about 4 days ago

chowman3, 2020-08-03:
it worked! thanks to everyone that helped

diogo5072, 2022-01-19:
were is tablet.cfg?
cuz i have the same prob

chowman3, 2022-01-19:
hi, unfortunately I stopped using my tablet a year ago, so I have no idea where it is.