hawku/TabletDriver #750: Last update 2 years ago

AuracleTech, 2020-08-02:
Where can I get in contact with the creator?
I can't find any information anywhere

ssccoonniicc, 2020-08-02:
Hawku basically just went radio silent, I have been trying to find a way to contact him but he probably doesn't want to update the driver anymore.

hawku, 2020-08-02:
@ssccoonniicc  Yeah, I don't really haven't had time or interest for this project.
There haven't really been any tablet hardware improvements in years, so I don't see a point to provide free marketing and support for shitty tablet manufacturers.
Quality of a reasonably priced tablet has only been dropping after I released the driver.

InfinityGhost, 2020-08-02:
Manufacturers are completely out of touch for what they're trying to market towards.

AuracleTech, 2020-08-02:
I contacted the Wacom company and told them about the Osu community and gaming community in general and the market there is. The person received the email replied with "We provide tablets for artists not for gamers." and did not transfer my email.
Unprofessional.
@hawku If you lose interest in this project could you at least define some users with rights to request pulls and stuff, there's a lot that could be going on at least. Might as well open a discord so everyone can communicate and self sustain, this would be quick and very nice!

InfinityGhost, 2020-08-02:
I used to maintain it with [my fork](https://github.com/InfinityGhost/TabletDriver) (now archived in favor of [my own cross-platform driver](https://github.com/InfinityGhost/OpenTabletDriver)), however ran into a major issue with every build being detected as malware on like 47 detection engines despite nothing of the such. IIRC, that occurs directly on `master` of this repo as well. Whether that be code that some malware used code that's identical to some in the GUI, such as the display screenshot querying, it became really annoying to cherrypick back until something else triggered it.

AuracleTech, 2020-08-03:
I really want to make one from scratch with this one as reference, not sure if I can do it tho.

ssccoonniicc, 2020-08-06:
Make it from scratch please