hawku/TabletDriver #755: Tablet Driver is not working for Star g430S

Veloctyy, 2020-08-06:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: Star G430S

When i try to use the tablet drivers its says its not connected like everyone else's problem. Also when i exit out of the driver it says "Some files from TabletDriverV0.2.3.zip were modified or new files were created. Do you wish to put them in archive?"
and i  downloaded vmulti and installed all Pentablet drivers and folders
[image]

Please help

ssccoonniicc, 2020-08-06:
Extract TabletDriver, don't use it from inside the zip file and add the g430s_B config if needed: https://www.reddit.com/r/osugame/comments/dnwtdk/potential_fix_for_xppen_g430s_and_community/. Also make sure you dont have any xp-pen drivers installed.

Veloctyy, 2020-08-07:
@ssccoonniicc when i open the drivers (and its not zip) its says the vmulti drivers wont open 
[image]

ssccoonniicc, 2020-08-08:
install vmulti and try to run TabletDriver as administrator

Veloctyy, 2020-08-08:
i did multiple times i even installed 64 and 32 bit and the regular vmulti

Veloctyy, 2020-10-16:
Ok so, xp pen made it so you can only use their drivers. So if you use an xp pen you should use this for the driver 
[image]
I hope this helped anyone who needed it. :)

closed 2020-10-16 by Veloctyy