hawku/TabletDriver #756: Tablet init failed! Xp-Pen G640

IGuiqik, 2020-08-06:
## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model: Xp-Pen G640

[2020-08-06 19:47:29.109] >> HIDTablet 0x056a 0x0374 0xFF0D 0x0001
[2020-08-06 19:47:29.110] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0374 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-08-06 19:47:29.110] >> Name "Wacom CTL-4100"
[2020-08-06 19:47:29.110] >> ReportId 0x10
[2020-08-06 19:47:29.110] >> ReportLength 192
[2020-08-06 19:47:29.110] >> DetectMask 0x20
[2020-08-06 19:47:29.110] >> MaxX 15200
[2020-08-06 19:47:29.110] >> MaxY 9500
[2020-08-06 19:47:29.110] >> MaxPressure 4095
[2020-08-06 19:47:29.110] >> Width 152.0
[2020-08-06 19:47:29.110] >> Height 95.0
[2020-08-06 19:47:29.110] >> TabletFormat WacomIntuosV3
[2020-08-06 19:47:29.110] >> AuxHID 0x056a 0x0376 0xFF0D 0x0001
[2020-08-06 19:47:29.110] >> AuxHID 0x056a 0x0374 0xFF0D 0x0001
[2020-08-06 19:47:29.110] >> AuxReportId 0x11
[2020-08-06 19:47:29.110] >> AuxReportLength 192
[2020-08-06 19:47:29.110] >> AuxCustomData ReportId Source=0
[2020-08-06 19:47:29.110] >> AuxCustomData ButtonsLow Source=1 SourceMask=0x0F
[2020-08-06 19:47:29.110] >> AuxButtonCount 4
[2020-08-06 19:47:29.111] >> #
[2020-08-06 19:47:29.111] >> # Wacom CTL-4100 Bluetooth
[2020-08-06 19:47:29.111] >> #
[2020-08-06 19:47:29.111] >> HIDTablet 0x056a 0x0377 0xFF0D 0x0001
[2020-08-06 19:47:29.112] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0377 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-08-06 19:47:29.112] >> Name "Wacom CTL-4100 Bluetooth (Experimental)"
[2020-08-06 19:47:29.112] >> ReportId 0x81
[2020-08-06 19:47:29.112] >> ReportLength 361
[2020-08-06 19:47:29.112] >> DetectMask 0xE0
[2020-08-06 19:47:29.112] >> MaxX 15200
[2020-08-06 19:47:29.112] >> MaxY 9500
[2020-08-06 19:47:29.112] >> MaxPressure 4095
[2020-08-06 19:47:29.112] >> Width 152.0
[2020-08-06 19:47:29.112] >> Height 95.0
[2020-08-06 19:47:29.112] >> #
[2020-08-06 19:47:29.112] >> # Wacom PTH-451 ???
[2020-08-06 19:47:29.112] >> #
[2020-08-06 19:47:29.112] >> HIDTablet 0x056a 0x0314 0xFF0D 0x01
[2020-08-06 19:47:29.114] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0314 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-08-06 19:47:29.114] >> Name "Wacom PTH-451"
[2020-08-06 19:47:29.114] >> ReportLength 10
[2020-08-06 19:47:29.114] >> DetectMask 0xE0
[2020-08-06 19:47:29.114] >> MaxX 31496
[2020-08-06 19:47:29.114] >> MaxY 19685
[2020-08-06 19:47:29.114] >> MaxPressure 2047
[2020-08-06 19:47:29.114] >> Width 157.0
[2020-08-06 19:47:29.114] >> Height 98.0
[2020-08-06 19:47:29.114] >> InitFeature 0x02 0x02
[2020-08-06 19:47:29.114] >> TabletFormat WacomIntuosV2
[2020-08-06 19:47:29.114] >> #
[2020-08-06 19:47:29.114] >> # Wacom PTH-850
[2020-08-06 19:47:29.114] >> #
[2020-08-06 19:47:29.114] >> HIDTablet 0x056A 0x0028 0x00D 0x0001
[2020-08-06 19:47:29.115] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0028 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-08-06 19:47:29.115] >> Name "Wacom PTH-850"
[2020-08-06 19:47:29.115] >> ReportLength 10
[2020-08-06 19:47:29.115] >> DetectMask 0xE0
[2020-08-06 19:47:29.115] >> MaxX 65024
[2020-08-06 19:47:29.115] >> MaxY 40640
[2020-08-06 19:47:29.115] >> MaxPressure 2047
[2020-08-06 19:47:29.115] >> Width 325.120
[2020-08-06 19:47:29.115] >> Height 203.200
[2020-08-06 19:47:29.115] >> InitFeature 0x02 0x02
[2020-08-06 19:47:29.115] >> TabletFormat WacomIntuosV2
[... 214 lines cut ...]
[2020-08-06 19:47:29.128]   Max Pressure = 8191
[2020-08-06 19:47:29.128]   Click Pressure = 0
[2020-08-06 19:47:29.128]   Keep Tip Down = 0 reports
[2020-08-06 19:47:29.128]   Report Id = 0x02
[2020-08-06 19:47:29.128]   Report Length = 12 bytes
[2020-08-06 19:47:29.128]   Detect Mask = 0x80
[2020-08-06 19:47:29.128]   Ignore Mask = 0x00
[2020-08-06 19:47:29.128]   Pen button map = 'MOUSE1' '' '' 
[2020-08-06 19:47:29.128]   Aux button count = 0
[2020-08-06 19:47:29.128]   Init report: { 0x02, 0xb0, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00 }
[2020-08-06 19:47:29.128] 
[2020-08-06 19:47:29.128] Area:
[2020-08-06 19:47:29.128]   Desktop = 1920px x 1080px
[2020-08-06 19:47:29.128]   Screen Map = 160px x 27px @ X-32000px, Y-32000px
[2020-08-06 19:47:29.128]   Tablet =  100.00mm x 50.00mm @ Min(50.00mm, 25.00mm) Max(150.00mm, 75.00mm)
[2020-08-06 19:47:29.128]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2020-08-06 19:47:29.128] 
[2020-08-06 19:47:29.128] >> Start
[2020-08-06 19:47:29.128] [ERROR] Tablet init failed!
[2020-08-06 19:47:29.128] [ERROR] Possible fixes:
[2020-08-06 19:47:29.128] [ERROR] 1) Uninstall other tablet drivers.
[2020-08-06 19:47:29.128] [ERROR] 2) Stop other tablet driver services.
Driver stopped!
 
I dont have any other drivers on my pc...

Acro13, 2020-08-06:
I'm getting the same issue. I bought my G640 2 days ago and my computer isn't, 1, picking up the tablet and 2 when I do have the tablet plugged in the osu drivers don't work. I've tried various different methods. such as having the XP-pen drivers installed which creates the result below. Or not having the XP-pen drivers install and getting no device connected prompt
[image]

Acro13, 2020-08-10:
Hello i found a solution

Go to the tablet drivers folder 
>Config
>tablet.cfg open in a file editor such as Notepad++
> find XP-Pen G640
> replace code with HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
#CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

>save changed
> restart drivers 
>works!

Aiko-bop, 2020-08-11:
@Acro13 Not sure how the fuck you figure that out but like.... You are dead ass sex on a stick, thanks amigo.