hawku/TabletDriver #764: Driver doesn't open and doesn't even show up on task manager

PoplessRebel, 2020-08-15:
## Hardware/software information
Windows Version: Microsoft Windows 10 Pro

Tablet Driver version: TabletdriverguiV2.3

Tablet model: Wacom CTL 4100

Ive been using the driver for the last year and out of nowhere it stopped working. Every time i open it, it shows up on the side icons and the moment i hover on the icon of the driver it will vanish. (Driver does NOT open no UI or anything)
I'm not sure if this is related to the problem but in the TabletDriverService.exe it says 
[image]

The original driver for my tablet (Wacom) works perfectly.
Tried reinstalling, using older versions, running as administrator, taking out the init.cfg file and running the program,running the program while the tablet is not plugged in, installing the program on a different drive ect and i am totally lost and don't know what to do so if there is anything i can do it will help a bunch.
Thank you.

Hyenook, 2020-08-17:
[Try InfinityGhost's TabletDriver](https://github.com/InfinityGhost/TabletDriver), or [OpenTabletDriver](https://github.com/InfinityGhost/OpenTabletDriver)

PoplessRebel, 2020-08-17:
> Try [InfinityGhost's TabletDriver](https://github.com/InfinityGhost/TabletDriver), or [OpenTabletDriver](https://github.com/InfinityGhost/OpenTabletDriver)

Opendriver does the same thing but it does open up the ui and before everything loads it closes back to the updater window and does the same vanish after hovering on it in the side icons thing.
and the tablet driver shows up as a virus

Hyenook, 2020-08-18:
OpenTabletDriver does work with the 4100, it was added in the last update. If you can't get it to work, you can join the Discord server for support.