hawku/TabletDriver #783: won't work with StarG430s (XP Pen)

iurns1x, 2020-08-24:
## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Hyenook, 2020-08-25:
You probably have the B revision of the G430S, try using InfinityGhost's fork of TabletDriver

iurns1x, 2020-08-25:
> You probably have the B revision of the G430S, try using InfinityGhost's fork of TabletDriver

i may be dumb but i downloaded the files and don’t know how to work it

Hyenook, 2020-08-25:
What's wrong? Extract it somewhere else, and run **TabletDriverGUI.exe**, and your tablet should be working. Unless you have downloaded the wrong file. Just in case, [I'll link the release page again.](https://github.com/InfinityGhost/TabletDriver/releases/tag/v0.3.1a)

spekjke, 2020-08-26:
i also have this problem with StarG430s (XP Pen) and i cant understand how to install  InfinityGhost's fork, there is others type of files

Hyenook, 2020-08-26:
This is going to get a bit long, so I'll not write how to get InfinityGhost's TabletDriver working (even though it's the same as Hawku's, I'm pretty sure your antivirus is just deleting all the files here)

Just add the following lines to the tablet.cfg file: 

`#
# XP-Pen G430S_B
#
HIDTablet 0x28BD 0x0913 0xFF0A 0x0001
CheckString Product "4 inch PenTablet"
Name "XP-Pen G430S_B"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 10160
MaxY 7620
MaxPressure 8191
Width 101.6
Height 76.2
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00`

spekjke, 2020-08-27:
thak a lot, now it works

spekjke, 2020-08-27:
thanks a lot, now it works

Aena1, 2020-09-02:
I also have a xp pen g430s and adding the following lines did not fix the problem