hawku/TabletDriver #788: my mouse is stuck in one place when i use antichatter

pastpast, 2020-08-27:
## Hardware/software information
Windows Version: windows 10

Tablet Driver version: v0.1.5.5 Devocub Edition with Antichatter

Tablet model/Mouse model: xp-pen g640/logitech g pro wireless

Hyenook, 2020-08-27:
When did you buy your tablet? Is it the Rev A or Rev B?

pastpast, 2020-08-27:
its the older one i think bought it over a year ago

Hyenook, 2020-08-28:
If it gets stuck in-game, then disabling Raw Input might help

DeoxyShocK, 2020-09-02:
In your tablet directory, open the 'config' folder and open 'tablet.cfg' in a text editor. Scroll down and find your tablet, or just ctrl+f your tablet, find the line that says 'DetectMask 0x80' and change it to 'DetectMask 0x20'. It should look like this when you're done: https://i.imgur.com/RPMQdBW.png

hongli-95, 2020-11-08:
> In your tablet directory, open the 'config' folder and open 'tablet.cfg' in a text editor. Scroll down and find your tablet, or just ctrl+f your tablet, find the line that says 'DetectMask 0x80' and [...]

This helped me solved the problem. Thanks.