hawku/TabletDriver #789: Tablet not found! G640

execeval, 2020-08-29:
I switch from g430S to g640.
I reinstall vmilti driver and hawku, uninstall all xp-pen drivers and anything related and check xp-pen services
When i open GUI i getting message that Driver Stopped 

```
[2020-08-28 13:01:53.500] >> CheckTablet
[2020-08-28 13:01:53.500] [ERROR] Tablet not found!
[2020-08-28 13:01:53.500] [ERROR] Check the list of supported tablets from the GitHub page.
[2020-08-28 13:01:53.500] [ERROR] http://github.com/hawku/TabletDrive
```

Here is my [startuplog.txt](https://github.com/hawku/TabletDriver/files/5145055/startuplog.txt)

## Hardware/software information
Windows Version: 7 SP1 X64

Tablet Driver version: TabletDriverV0.2.3

Tablet model: Xp-pen g640

execeval, 2020-08-29:
This comment from the previous issue helped me.
[https://github.com/hawku/TabletDriver/issues/746#issuecomment-665738653](url)

Please don't close issue until this problem resolved automatically

Hyenook, 2020-08-30:
Actually close the issue, this is never going to get closed since Hawku stopped the development of the driver

elmerr100, 2020-08-30:
https://www.youtube.com/watch?v=m2aJDX5730I

Here, this might help.

They released a new version of the G640. It most probably won't work for the new version of it.
Did you just purchase the tablet recently?

OsuNoob, 2020-09-10:
I had this same problem and it is probably because you got the rev b and you bought it recently 
https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/
this website is what I found when I was looking for the fix you just have to copy the pastebin and change your tablet.cfg to what you copied i hope this helps