hawku/TabletDriver #790: Can you guys add support for the XP-Pen Star 03? (not the v2 one)

elmerr100, 2020-08-30:
## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

ssccoonniicc, 2020-08-30:
You will probably need to make the config yourself, check: https://github.com/InfinityGhost/TabletDriver/wiki/Creating-Tablet-Configurations

elmerr100, 2020-08-31:
This helped me alot, but how do I get the ReportID, the DetectMask and the InitReport?

elmerr100, 2020-08-31:
> You will probably need to make the config yourself, check: https://github.com/InfinityGhost/TabletDriver/wiki/Creating-Tablet-Configurations

It keeps telling me "Can't open" Is that because the config is wrong or the driver just really doesn't support it. By the way i'm still not sure if the ReportID, DetectMask, and the InitReport I put was correct.

elmerr100, 2020-08-31:
@ssccoonniicc I need a little help if you could lend me any. I can't make it right. Thanks.

ssccoonniicc, 2020-08-31:
Your tablet may require an entirely new report parser but I really don't know. You can try asking for help in the otd discord though: discord.gg/p3gydus (just @InfinityGhost and say you need help creating a config for your tablet in TabletDriver)