hawku/TabletDriver #792: Driver stopped. Restarting! Check Console !!!

TitoBre, 2020-08-30:
## Hardware/software information
Windows Version: 10 

Tablet Driver version: V0.2.3

Tablet model: Wacom cth 480
[image]

TitoBre, 2020-08-30:
btw I uninstalled all my tablet drivers. I updated my PC. I reinstalled the Tablet Driver. I did everything and now idk what to do to make the driver work. :c

hydra1489, 2020-08-31:
this same thing happened to me. i use the xp pen star g640 (or at least I'm trying to) I uninstalled the xp driver and got the hawku ones as a friend told me to. once i started the driver it worked for a second and now whatever i change my area to the sensitivity stays the same. and when i try to delete the folder or move it it says "the action can't be completed because the folder or file is open in another program" the driver won't restart either it doesn't even detect my tablet or so it says. and to top it all off when i go into Osu! the cursor Immediately goes to the top left corner and can't be moved unless i use a mouse.

Hyenook, 2020-08-31:
That's another driver running in the background, you might have to use third-party tools to properly uninstall the driver, like Revo Uninstaller (or any other open or closed-source tool)

Hyenook, 2020-08-31:
Btw, @TitoBre since you have a Wacom tablet you can use OpenTabletDriver instead

TitoBre, 2020-08-31:
@Hyenook Were can I install the OpenTabletDriver? U got a link?

Hyenook, 2020-08-31:
https://github.com/InfinityGhost/OpenTabletDriver There's also a Discord link there if you need help with installing the driver.

closed 2020-08-31 by TitoBre