hawku/TabletDriver #793: virus detected

donertfs, 2020-08-31:
i can't download the hawku drivers because chrome detects a virus. I have already turned off the virus protection on chrome, and no other browser lets me download it.

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Hyenook, 2020-09-01:
You're probably downloading InfinityGhost's fork of TabletDriver. [Here a DDL link for Hawku's TabletDriver.](https://github.com/hawku/TabletDriver/releases/download/v0.2.3/TabletDriverV0.2.3.zip)

donertfs, 2020-09-01:
> You're probably downloading InfinityGhost's fork of TabletDriver. [Here a DDL link for Hawku's TabletDriver.](https://github.com/hawku/TabletDriver/releases/download/v0.2.3/TabletDriverV0.2.3.zip)

Thank you, that worked!

donertfs, 2020-09-01:
> You're probably downloading InfinityGhost's fork of TabletDriver. [Here a DDL link for Hawku's TabletDriver.](https://github.com/hawku/TabletDriver/releases/download/v0.2.3/TabletDriverV0.2.3.zip)

Thank you, i was actually able to download that, but do you have the 0.3.1a version that has gaomon s620 support?

Hyenook, 2020-09-02:
> do you have the 0.3.1a version that has gaomon s620 support?
That's the one you were trying to download. But since for some reason you aren't unable to disable your antivirus, you can manually add the config for it.

Add the following lines to the tablet.cfg file

`#
# Gaomon S620
#
HIDTablet 0x256C 0x006D 0xFF00 0x0001
CheckString 201 "OEM02_T18e"
Name "Gaomon S620"
ReportId 0x00
ReportLength 16
DetectMask 0x80
MaxX 33020
MaxY 20320
MaxPressure 8191
Width 165.1
Height 101.6
InitStrings 100 200
InitStrings 100 200
AuxReportId 0x08
AuxReportLength 12
AuxDetectMask 0xE0
AuxgnoreMask 0x80
AuxCustomData ReportId Source=0
AuxCustomData ButtonsLow Source=4 SourceMask=0x3F
AuxCustomData Detect Source=1
AuxButtonCount 4`