hawku/TabletDriver #802: Cursor TP and disappear

MalwarePup, 2020-09-09:
Hello,

I have recently bought a Wacom CTH-480 but when I want to play with her, my cursor full TP and disappear.I have changed the original cable by a RAMPOW 2 meters. The original is too short but with him no problems..

But with the original cable and my 2 RAMPOW, whenever the cursor stop move and he his blocked on the screen. And whenever she works again.

Check my video : https://www.youtube.com/watch?v=JcWiH7JsBZ0&feature=youtu.be

## Hardware/software information
Windows Version: W10 pro x64

Tablet Driver version: 0.2.3

Tablet model: CTH-480-S

Hyenook, 2020-09-10:
This happens because there's another driver running in the background. You could try fixing this by installing the default Wacom driver, and uninstalling it.

MalwarePup, 2020-09-10:
I have tried before but the same result.

MalwarePup, 2020-09-10:
I have foud the issue.
I put my CTH-480 on the logitech powerplay and let my G pro on it too. But le powerplay create interference with the tablet.
So i just push the powerplay on my desk and work.

I have another question, when i play with my mouse, the animation of the cursor is so smooth but when i play with the tablet, she has 500Hz and the cursor have bad animation and slowly versus mouse.

Hyenook, 2020-09-10:
Does the same thing happens when you disable the smoothing filter?

MalwarePup, 2020-09-10:
I have disable the smoothing filter and this is very better, thanks for help.

closed 2020-09-10 by MalwarePup