hawku/TabletDriver #809: driver 0.1.5.5 antichatter error with 1 ms of latency to get the "smooth"

taifokk, 2020-09-15:
Why my cursor feels laggy when i have 1 ms of latency to get the "smooth" , and obviously the smooth filter ON, and when i put the number 0 instead of 1 in ms of latency , my cursor feels normal/responsive , but the shakes come back ... , a few days ago my driver works perfectly :( pls help

## Hardware/software information
Windows Version: windows 10

Tablet Driver version: 0.1.5.5 devocub edition

Tablet model: ctl 480

ssccoonniicc, 2020-09-15:
Did you update windows a few days ago?

taifokk, 2020-09-16:
yes i updated windows to 2004 version

ssccoonniicc, 2020-09-16:
Windows broke a function, use https://www.github.com/AbdShullah/TabletDriver or https://www.github.com/pizzabelly/TabletDriver