hawku/TabletDriver #815: XP-Pen G640 Driver stopped. Restarting! Check console !!!

Cramelly, 2020-09-20:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: V0.2.3

Tablet model: XP-Pen G640

I have just got a new PC with a fresh installation of windows, I installed the hawku driver for my XP-Pen G640. Whenever I try to restart the driver it says "Driver stopped. Restarting! Check console !!!" I checked the console and it just repeats the same thing over again with mention of "Wacom CTH-470 (wacom drivers Installed)"
[image]

DeadHunterX20, 2020-09-21:
I had the same problem and the thing i did helped me so i will tell you what i did.
1.Open Tablet Driver Gui then connect your tablet through usb to pc 
2. Now open Device manager on your windows and delete all drivers that are from your tablet 
3.After that reinstall  hawku/TabletDriver, just go to the folder and click on "install_vmulti_driver.bat"
4. Now disconnect the tablet and connect it again, it should work now

Cramelly, 2020-09-22:
Thanks it is working again

closed 2020-09-22 by Cramelly