hawku/TabletDriver #820: CTL 480 problem ( cant connect) it never stop

osuguys, 2020-09-27:
[2020-09-27 20:24:38.387] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2020-09-27 20:24:38.390] [ERROR] Can't open VMulti device!

[2020-09-27 20:24:38.390] [ERROR] Possible fixes:
[2020-09-27 20:24:38.390] [ERROR] 1) Install VMulti driver
[2020-09-27 20:24:38.390] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2020-09-27 20:24:38.390] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2020-09-27 20:24:43.953] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2020-09-27 20:24:43.956] [ERROR] Can't open VMulti device!

[2020-09-27 20:24:43.956] [ERROR] Possible fixes:
[2020-09-27 20:24:43.956] [ERROR] 1) Install VMulti driver
[2020-09-27 20:24:43.956] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2020-09-27 20:24:43.956] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2020-09-27 20:24:49.507] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2020-09-27 20:24:49.511] [ERROR] Can't open VMulti device!

[2020-09-27 20:24:49.511] [ERROR] Possible fixes:
[2020-09-27 20:24:49.511] [ERROR] 1) Install VMulti driver
[2020-09-27 20:24:49.511] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2020-09-27 20:24:49.511] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model: ctl 480

osuguys, 2020-09-27:
help because i wann play osui

ssccoonniicc, 2020-09-29:
Try running TabletDriver as administrator and uninstall any wacom drivers you have

awevxz, 2020-11-11:
Hey, idk if this will but it works for me:)

If you turn on run at windows startup in the GUI and keep your tablet physically connected, shut down your pc and then turn it back on, don't log in until it's completely booted up, I wait like at least 2 minutes just to make sure it has booted fully (I have a shit pc btw). The driver should connect to your tablet automatically and start working. I don't know if this will work for anybody else but just make sure you've installed vmulti_drivers and that your tablet is supported.

I'm using the Wacom CTL-4100 and running windows 10.

osuguys, 2020-11-11:
Hi awevxz

I'm done that 2 time but it still a same thing 

I think i cant download the vmulti_driver

awevxz, 2020-11-11:
Have you installed the microsoft framework?

osuguys, 2020-11-11:
Have you installed the microsoft framework? 

yes