hawku/TabletDriver #824: Lag with smoothing filter enabled recently

obeliskosu, 2020-09-29:
Pretty sure after this windows feature update recently something messed it up if i put any other number apart from 0 with smoothing filter enabled my cursor when dragging or close to tablet is laggy n so messed up idk what to do any ideas? 

## Hardware/software information
Windows Version: 10

Tablet Driver version: devocubs latest

Tablet model: wacom cth 480

ssccoonniicc, 2020-09-29:
use https://www.github.com/AbdShullah/TabletDriver or https://www.github.com/pizzabelly/TabletDriver

obeliskosu, 2020-09-29:
ty ill try it out later tonight hopefully it works

inyafaze, 2020-12-07:
> use https://www.github.com/AbdShullah/TabletDriver or https://www.github.com/pizzabelly/TabletDriver

doesn't work for me..