hawku/TabletDriver #827: XP-PEN G430S Not working

SzukiX, 2020-09-30:
Hey i just installed the drivers for it and its keep saying Driver stopped. Restarting! Check console !!! ive uninstalled the xppen drivers restarted uninstalled and reinstalled everything and its not working here is my console

[2020-09-30 16:44:25.229] >> DetectMask 0x40
[2020-09-30 16:44:25.229] >> MaxX 21600
[2020-09-30 16:44:25.229] >> MaxY 13500
[2020-09-30 16:44:25.229] >> MaxPressure 1023
[2020-09-30 16:44:25.229] >> Width 216.000
[2020-09-30 16:44:25.229] >> Height 135.000
[2020-09-30 16:44:25.229] >> InitFeature 0x02 0x02
[2020-09-30 16:44:25.229] >> # Wacom CTH-460
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D1 0xFF0D 0x0001
[2020-09-30 16:44:25.229] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D1 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D1 0x000D 0x0001
[2020-09-30 16:44:25.229] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D1 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.229] >> Name "Wacom CTH-460"
[2020-09-30 16:44:25.229] >> ReportLength 9
[2020-09-30 16:44:25.229] >> DetectMask 0x40
[2020-09-30 16:44:25.229] >> MaxX 14720
[2020-09-30 16:44:25.229] >> MaxY 9200
[2020-09-30 16:44:25.229] >> MaxPressure 1023
[2020-09-30 16:44:25.229] >> Width 147.200
[2020-09-30 16:44:25.229] >> Height 92.000
[2020-09-30 16:44:25.229] >> InitFeature 0x02 0x02
[2020-09-30 16:44:25.229] >> # Wacom CTH-460(A)
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D6 0xFF0D 0x0001
[2020-09-30 16:44:25.229] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D6 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D6 0x000D 0x0001
[2020-09-30 16:44:25.229] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D6 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.229] >> Name "Wacom CTH-460(A)"
[2020-09-30 16:44:25.229] >> ReportLength 9
[2020-09-30 16:44:25.229] >> DetectMask 0x40
[2020-09-30 16:44:25.229] >> MaxX 14720
[2020-09-30 16:44:25.229] >> MaxY 9200
[2020-09-30 16:44:25.229] >> MaxPressure 1023
[2020-09-30 16:44:25.229] >> Width 147.200
[2020-09-30 16:44:25.229] >> Height 92.000
[2020-09-30 16:44:25.229] >> InitFeature 0x02 0x02
[2020-09-30 16:44:25.229] >> # Wacom CTH-461
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D2 0xFF0D 0x0001
[2020-09-30 16:44:25.229] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D2 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D2 0x000D 0x0001
[2020-09-30 16:44:25.229] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D2 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.229] >> Name "Wacom CTH-461"
[2020-09-30 16:44:25.229] >> ReportLength 9
[2020-09-30 16:44:25.229] >> DetectMask 0x40
[2020-09-30 16:44:25.229] >> MaxX 14720
[2020-09-30 16:44:25.229] >> MaxY 9200
[2020-09-30 16:44:25.229] >> MaxPressure 1023
[2020-09-30 16:44:25.229] >> Width 147.200
[2020-09-30 16:44:25.229] >> Height 92.000
[2020-09-30 16:44:25.229] >> InitFeature 0x02 0x02
[2020-09-30 16:44:25.229] >> # Wacom CTH-461(A)
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D7 0xFF0D 0x0001
[2020-09-30 16:44:25.229] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D7 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.229] >> HIDTablet 0x056A 0x00D7 0x000D 0x0001
[2020-09-30 16:44:25.244] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D7 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.244] >> Name "Wacom CTH-461(A)"
[2020-09-30 16:44:25.244] >> ReportLength 9
[2020-09-30 16:44:25.244] >> DetectMask 0x40
[2020-09-30 16:44:25.244] >> MaxX 14720
[2020-09-30 16:44:25.244] >> MaxY 9200
[2020-09-30 16:44:25.244] >> MaxPressure 1023
[2020-09-30 16:44:25.244] >> Width 147.200
[2020-09-30 16:44:25.244] >> Height 92.000
[2020-09-30 16:44:25.244] >> InitFeature 0x02 0x02
[2020-09-30 16:44:25.244] >> # Wacom CTH-461SE
[2020-09-30 16:44:25.244] >> HIDTablet 0x056A 0x00DA 0xFF0D 0x0001
[2020-09-30 16:44:25.244] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DA UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-09-30 16:44:25.244] >> HIDTablet 0x056A 0x00DA 0x000D 0x0001
[2020-09-30 16:44:25.244] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DA UsagePage=0x000D Usage=0x0001 Exclusive=False
[... 209 lines cut ...]
[2020-09-30 16:44:25.282] [STATUS] COMMANDS CheckDeviceString CheckString CheckTablet ClearAuxButtonMap ClearAuxCustomData ClearButtonMap 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS ClearCustomData ClickPressure CloseTablet CustomDataInstruction AuxCustomData CustomData 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS Debug DesktopSize Desktop DetectMask Echo Exit Quit FilterTester Tester FilterTimerInterval Interval TimerInterval 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS GetCommands GetDeviceStrings GetString GetStrings GetFeature GravityFilter Gravity 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS Height Help HIDList HIDTablet IgnoreMask Include Information Info InitFeature InitReport 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS InitStrings InitString KeepTipDown ListCommands List ListKeys Log LogDirect LogTabletArea 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS MaxPressure MaxX MaxY Measure NoiseReduction Noise Output OutputMode Mode OutputReport Report 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS PressureDeadzone PressureSensitivity RelativeResetDistance ResetDistance RelativeResetTime ResetTime 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS RelativeSensitivity Sensitivity ReportId ReportLength Rotate ScreenArea ScrollAcceleration ScrollAcc 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS ScrollSensitivity ScrollStopCursor SetFeature Feature Skew SmoothingFilter Smoothing 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS Start Status TabletArea Area TabletDataFormat Format TabletFormat TabletMove Move 
[2020-09-30 16:44:25.282] [STATUS] COMMANDS TabletName Name TabletValid USBTablet USBWrite Wait Width 
[2020-09-30 16:44:25.282] 
[2020-09-30 16:44:25.282]   Driver version: 0.2.3
[2020-09-30 16:44:25.282]   Windows version: Microsoft Windows NT 6.2.9200.0
[2020-09-30 16:44:25.282]   Windows product: Windows 10 Enterprise
[2020-09-30 16:44:25.282]   Windows release: 2004
[2020-09-30 16:44:25.282] 
[2020-09-30 16:44:25.282] >> CheckTablet
[2020-09-30 16:44:25.282] [ERROR] Tablet not found!
[2020-09-30 16:44:25.282] [ERROR] Check the list of supported tablets from the GitHub page.
[2020-09-30 16:44:25.282] [ERROR] http://github.com/hawku/TabletDriver
Driver stopped. Restarting!

any help please thx!

SzukiX, 2020-09-30:
for got to add

Windows Version: Windows 10 64Bit

Tablet Driver version: V0.2.3

Tablet model: XP-PEN G430S

Aena1, 2020-10-02:
I am having the same issue with the xp pen g430s on hawku.

ssccoonniicc, 2020-10-02:
https://www.reddit.com/r/osugame/comments/dnwtdk/potential_fix_for_xppen_g430s_and_community

Veloctyy, 2020-10-16:
Ok so, xp pen made it so you can only use their drivers. So if you use an xp pen you should use this for the driver
I hope this helped anyone who needed it. :)
[image]

ssccoonniicc, 2020-10-18:
You can still use the g430s with hawku's driver, you just need to add the config.