hawku/TabletDriver #834: Driver stopped! Tablet not found

kaspishehe, 2020-10-07:
## Hardware/software information
Windows Version:  10.0.19041 Build 19041

Tablet Driver version: your latest released driver

Tablet model:  xp pen g640

Hello, i have problem with my xp pen g640. Tablet not found and driver stopped.
I installed vmulti and uninstalled all previous drivers, still doesn't work.

[vmulti log.txt](https://github.com/hawku/TabletDriver/files/5340655/vmulti.log.txt)
[startuplog.txt](https://github.com/hawku/TabletDriver/files/5340657/startuplog.txt)

ThePandaxXx, 2020-10-11:
Got the same exact one recently and trying a bunch of methods to try to fix it.

Veloctyy, 2020-10-16:
Ok so, xp pen made it so you can only use their drivers. So if you use an xp pen you should use this for the driver
I hope this helped anyone who needed it. :)
[image]

kaspishehe, 2020-10-16:
@ThePandaxXx 

I already returned my tablet. Watch this below

https://www.reddit.com/r/osugame/comments/j6vn8p/hawku_drivers_doesnt_work/
https://youtu.be/m2aJDX5730I

awevxz, 2020-11-11:
Hey, idk if this will work for anybody else but it works for me:)

If you turn on run at windows startup in the GUI and keep your tablet physically connected, shut down your pc and then turn it back on, don't log in until it's completely booted up, I wait like at least 2 minutes just to make sure it has booted fully (I have a shit pc btw). The driver should connect to your tablet automatically and start working. I don't know if this will work for anybody else but just make sure you've installed vmulti_drivers and that your tablet is supported.

I'm using the Wacom CTL-4100 and running windows 10.