hawku/TabletDriver #844: Drivers stopped! Restarting! Check Console!!

AidanClicks, 2020-10-19:
Having problems with running the drivers properly. Tried everything requested with no positive feedback.

## Hardware/software information
Windows Version: 10. 64 bit

Tablet Driver version: 0.2.3

Tablet model: Veikk s640

ski7shredd, 2020-10-20:
I also have this issue, it started after I did a windows update (16.10) and I don't know the solution. Also in device manager it seems like driver is missing, when I plug the tablet, it shows nothing but when I unplug it, it shows a unreaded usb device. I also did a reinstall with tablet drivers, Huion H640P.

RikkaWls, 2020-11-04:
you should useinfinityghost driver i also have a huion h640p that driver works with my tablet
https://github.com/InfinityGhost/OpenTabletDriver

awevxz, 2020-11-11:
Hey, idk if this will work for anybody else but it works for me:)

If you turn on run at windows startup in the GUI and keep your tablet physically connected, shut down your pc and then turn it back on, don't log in until it's completely booted up, I wait like at least 2 minutes just to make sure it has booted fully (I have a shit pc btw). The driver should connect to your tablet automatically and start working. I don't know if this will work for anybody else but just make sure you've installed vmulti_drivers and that your tablet is supported.

I'm using the Wacom CTL-4100 and running windows 10.