hawku/TabletDriver #845: Can't use smoothing filter with the newest windows update

bchan023, 2020-10-21:
Ever since i updated windows, i'm just not able to use the smoothing filter. the drivers work completely fine without it, but once you turn on the filter, it gives it unbearable input lag and doesnt even track correctly.

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: CTL 460

belkan4, 2020-10-23:
I have the same issue , i installed the windows update today 
CTL 472
Devocub V0.1.5.5
Windows 10 Pro v2004

inyafaze, 2020-10-26:
Same issue but with the CTL-470 
it's been like that for some weeks now 
Devocub v0.1.5.5
Windows 10 Pro

xAeryn, 2020-11-01:
Same issue, i think it was win10 2004 update that broke the smoothing filter. My usual 15ms smoothing setting has given a huge amount of lag, i wanna say around +100ms, since then - that makes it no longer usable at all. I'm not sure why this would happen or if it's fixable on driver side, but please take a look.

bchan023, 2020-11-01:
I was on the regular drivers on version 0.2 for awhile, however switching to the Devocub 0.1.5.5 worked for me and I was able to use smoothing as usual.

xAeryn, 2020-11-02:
Ran the uninstall bat and then reinstalled from v0.2.3 and it still doesn't work, same behavior - but it may work if i reduce the smoothing filter delay to a tiny value (like less than 1 millisecond). Maybe something is wrong with the time scale so it thinks that 0.5ms is actually 10-20ms.