hawku/TabletDriver #852: Does the Star G430s work in the drivers? or do i have to download the xp pen drivers?

Eikkie, 2020-10-27:
I want to use the hawku tablet driver because the xp pen drivers have to make you guess the spots for it

## Hardware/software information
Windows Version: windows 7 ultimate

Tablet Driver version: 

Tablet model: Star G430s

Veloctyy, 2020-10-28:
xp pen made it so there tablet only works with there drivers so you have to use their drivers

SpaghetCat, 2020-10-31:
I have been using the Hawku drivers for a while now with my xp pen g430s and it works fine. The only problem i had was that it did not detect my tablet at first but you can put the text furder down in this comment in your **tablet.cfg** file. You can find the **tablet.cfg** in tablet driver folder> config> tablet.cfg

This text:
#
# XP-Pen G430S_B
#
HIDTablet 0x28BD 0x0913 0xFF0A 0x0001
CheckString Product "4 inch PenTablet"
Name "XP-Pen G430S_B"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 10160
MaxY 7620
MaxPressure 8191
Width 101.6
Height 76.2
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

Veloctyy, 2020-11-02:
> I have been using the Hawku drivers for a while now with my xp pen g430s and it works fine. The only problem i had was that it did not detect my tablet at first but you can put the text furder down in this comment in your **tablet.cfg** file. You can find the **tablet.cfg** in tablet driver folder> config> tablet.cfg
> 
> This text:
> 
> ```
> #
> # XP-Pen G430S_B
> #
> 
> HIDTablet 0x28BD 0x0913 0xFF0A 0x0001
> CheckString Product "4 inch PenTablet"
> Name "XP-Pen G430S_B"
> ReportId 0x02
> ReportLength 12
> DetectMask 0x80
> MaxX 10160
> MaxY 7620
> MaxPressure 8191
> Width 101.6
> Height 76.2
> InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
> ```
> 
> i got this from:
> #783

I tried this but it didn't work. It just said it couldn't download Vmulti.

SpaghetCat, 2020-11-02:
When did you buy the tablet? Because the old one is not supported

And have you tried installing the Vmulti driver by opening **install_driver_vmulti.bat**

Veloctyy, 2020-11-02:
> When did you buy the tablet? Because the old one is not supported
> 
> And have you tried installing the Vmulti driver by opening **install_driver_vmulti.bat**

I got my tablet in June of 2020 and I did install vmulti

SpaghetCat, 2020-11-02:
I am not an expert in these kind of things but the last thing that i can think of is reinstalling the vmulti driver
And if that doesnt work you should ask someone else i hope this will work but i cant guarantee it

Trol, 2020-11-20:
[image]
You might need to install this, not sure if it's required but it does say **huion** in the folder name and it's a "driver" of some sorts.