hawku/TabletDriver #862: Cursor stuck if raw input is enabled

ItsFeold, 2020-11-08:
I want to change sensitivy in game and want to enable raw input, and at first it works, but if i go to play like 2 hours later or something cursor just gets stuck. I cant exactly figure out when it starts going to shit

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: Wacom CTH-490

UltimateKaren, 2020-11-09:
You gotta turn raw input off. This happens a ton with most tablets and drivers, not just hawku

ItsFeold, 2020-11-09:
> You gotta turn raw input off. This happens a ton with most tablets and drivers, not just hawku
is there absolutely no way of having it on? It's so much more convienient this way

UltimateKaren, 2020-11-09:
The only other way is to disable windows ink, but I'm not sure if it works for everyone. It didn't work for me so I had to turn raw input off. Here's a thread with links on how to disable windows ink: https://osu.ppy.sh/community/forums/topics/590367