hawku/TabletDriver #882: XP-Pen G640 Driver stopped. restarting. PLS HELP what can i do? :"C

Gerike22, 2020-11-25:
## Hardware/software information
Windows Version: Win 10   

Tablet Driver version: 0.2.3

Tablet model: XP-Pen Star G640

[image]
 help pls :C

PqndaCheats, 2020-11-25:
Me pasa igual

Gerike22, 2020-11-25:
> Me pasa igual

hey bro i found a solution
go to the tablet.cfg
find the g640
and paste this
HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
#CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

PqndaCheats, 2020-11-28:
Its okay, i found another drivers and it works correctly, thx u