hawku/TabletDriver #940: [VMULTI] Tablet Init failed, Devcon.exe failed.

AWandererInInternet, 2020-12-28:
I saw a couple of issues like this, devcon.exe and tablet init failed, just caring to bring this up once again and hopefully this really gets resolved.

## Hardware/software information
Windows Version: Windows 10 20H2 64-Bit

Tablet Driver version: 0.2.3

Tablet model: XP-PEN STAR G640S

## Devcon.exe failed.

If I try to execute **install_vmulti_driver.bat** without running as admin, the bat does not request for admin permissions as expected and instead gave me this:
```
  Are you sure you want to install vmulti driver?

Press enter to install...
C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>REM  --> Check for permissions

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>"C:\WINDOWS\system32\cacls.exe" "C:\WINDOWS\system32\config\system"   1>nul 2>&1

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>REM --> If error flag set, we do not have admin.

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>if '0' NEQ '0' (
echo Requesting administrative privileges...
 goto UACPrompt
)  else (goto gotAdmin  )

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>if exist "C:\Users\C\AppData\Local\Temp\getadmin.vbs" (del "C:\Users\CAppData\Local\Temp\getadmin.vbs"  )

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>pushd "C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti"

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>CD /D "C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti\"

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>devcon.exe /r install vmulti.inf pentablet\hid
devcon.exe failed.
Press enter...
```
## Running install_vmulti_driver.bat as administrator
-- > When I run the bat **install_vmulti_driver.bat** as administrator, the bat just *immediately disappears* without any prompt after pressing **enter**.

However, going into the **driver_vmulti** folder where the 64 bit driver is located of course, executing the bat **install_vmulti_64.bat** without any admin permissions, it does not request or asks for admin permissions and instead gave me an outcome of **devcon.exe failed** similar above.
```
C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>REM  --> Check for permissions

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>"C:\WINDOWS\system32\cacls.exe" "C:\WINDOWS\system32\config\system"   1>nul 2>&1

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>REM --> If error flag set, we do not have admin.

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>if '0' NEQ '0' (
echo Requesting administrative privileges...
 goto UACPrompt
)  else (goto gotAdmin  )

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>if exist "C:\Users\C\AppData\Local\Temp\getadmin.vbs" (del "C:\Users\CAppData\Local\Temp\getadmin.vbs"  )

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>pushd "C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti"

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>CD /D "C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti\"

C:\Users\C\Desktop\TabletDriverV0.2.3\driver_vmulti>devcon.exe /r install vmulti.inf pentablet\hid
devcon.exe failed.
Press enter...
```

## How I resolved Devcon.exe failed

**But** if I run **install_vmulti_64.bat** from the driver_multi folder as administrator, the driver installs as expected.

```
C:\WINDOWS\system32>REM  --> Check for permissions
[... 197 lines cut ...]
[2020-12-28 14:18:43.964]   Max Y = 20000
[2020-12-28 14:18:43.964]   Max Pressure = 8191
[2020-12-28 14:18:43.964]   Click Pressure = 0
[2020-12-28 14:18:43.964]   Keep Tip Down = 0 reports
[2020-12-28 14:18:43.964]   Report Id = 0x02
[2020-12-28 14:18:43.964]   Report Length = 10 bytes
[2020-12-28 14:18:43.970]   Detect Mask = 0x80
[2020-12-28 14:18:43.970]   Ignore Mask = 0x00
[2020-12-28 14:18:43.970]   Pen button map = 'MOUSE1' 'MOUSE2' 'MOUSE3' 
[2020-12-28 14:18:43.970]   Aux button count = 0
[2020-12-28 14:18:43.970]   Init report: { 0x02, 0xb0, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00 }
[2020-12-28 14:18:43.970] 
[2020-12-28 14:18:43.970] Area:
[2020-12-28 14:18:43.970]   Desktop = 1024px x 768px
[2020-12-28 14:18:43.970]   Screen Map = 1024px x 768px @ X+0px, Y+0px
[2020-12-28 14:18:43.970]   Tablet =  66.67mm x 50.00mm @ Min(40.00mm, 25.00mm) Max(106.67mm, 75.00mm)
[2020-12-28 14:18:43.970]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2020-12-28 14:18:43.970] 
[2020-12-28 14:18:43.970] >> Start
[2020-12-28 14:18:43.970] [ERROR] Tablet init failed!
[2020-12-28 14:18:43.970] [ERROR] Possible fixes:
[2020-12-28 14:18:43.970] [ERROR] 1) Uninstall other tablet drivers.
[2020-12-28 14:18:43.970] [ERROR] 2) Stop other tablet driver services.
Driver stopped!
```

AWandererInInternet, 2020-12-28:
Update as of this time: After reinstalling Windows 10 with my files and data still intact, the Devcon.exe failed and the No UAC prompt is gone, so it must be a Windows Installation problem since I recently upgraded from 7 to 10, upgrade seems probably the conflict as it replaces files and stuff like that.

AWandererInInternet, 2020-12-29:
**Update as of this time**: After further investigation, about the *Devcon failed. issue*, it turns out to be a permissions problem, since I granted myself permissions to view system32 and the config folder inside system32 when I was doing something, I removed myself from the permissions list and everything works as it should now.

How to reproduce devcon.exe failed:
Grant yourself admin permissions to **view** the system32 folder and **config** folder.
Open the install or remove drivers hawku .bat files.

AWandererInInternet, 2020-12-30:
## devcon.exe failed problem not involving permissions
From what I see, their problem is not a folder permissions problem.
And the majority, their OS is Windows 8.
It goes like this:
```Device node created
Updating drivers for xxx
devcon.exe failed
```

This is entirely different from my issue shown above.

Xayanide, 2021-03-12:
## December 16, 2018 - Default Tablet Configuration of G640s Unchanged in TabletDriverV0.2.3
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00
```

## February 13, 2021 - Found a Solution for Tablet Init failed

Sources, Credits, References: 

Issue Number 
#990 
Method 1
https://github.com/hawku/TabletDriver/issues/990#issuecomment-778399795
Method 2, Method 3 
https://github.com/hawku/TabletDriver/issues/990#issuecomment-778567638

## Method 1

Find ```XP-Pen G640s``` in ```TabletDriverV0.2.3\config\tablet.cfg```
Change InitReport to 
```InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00```
Save changes and restart driver.

Method 1 Config:
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

## Method 2
If that method did not help you, in my case, it didn't, here is another method that did solved my issue, hopefully it works for you as well.

Change ReportLength to
```ReportLength 12```

Replace InitReport with 
```InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00```

Method 2 Config:
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

## Method 3
However, if it still didn't work for you and still gave you the error, try these below.

```ReportLength 10``` 
```InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x0```

Method 3 Config:
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x0
```

## Post Scriptum

You can copy paste the Config code blocks and paste it into your tablet.cfg file, find G640s section and paste it there.

Alternatively just change each ```ReportLength``` and ```InitReport``` since those two are the only ones needed changes.

I made a ZIP file, inside are folders with ```tablet.cfg``` files.
Retaining every configuration of tablets,
But only changed the G640s section.

Attachment:
[G640s Tablet Configs.zip](https://github.com/hawku/TabletDriver/files/6132257/G640s.Tablet.Configs.zip)

It has folders namely ```Method 1```, ```Method 2```, ```Method 3``` with ```tablet.cfg``` files. 
With different configurations being made according to the methods above, this is just an alternative if you are lazy copy pasting texts and just want to copy paste and replace the config folder and the files inside altogether.

Steps:
1 Open ```G640s.Tablet.Configs.zip``` or extract it & move to Step 3
2 Choose and Open any of the folders namely ```Method 1```, ```Method 2```, ```Method 3```
3 Copy or Move the ```config``` folder.
4 Paste it to your ```TabletDriverV0.2.3``` folder.
5 If it asks you to Replace the folder or ```tablet.cfg```
6 Replace it.
7 Run ```TabletDriverGUI.exe```
If the Error appears, try the other Method folders and do the steps again.

SpicyDice, 2021-03-12:
Method 2 Solved my Issue.
You have my thanks for gathering this information and putting it altogether.

Thanks also for providing the sources and crediting them.

SpicyDice, 2021-03-12:
You have my thanks for gathering these Information, putting them altogether, providing and crediting the sources.

SpicyDice, 2021-03-12:
You have my thanks for providing the sources, gathering the information and putting them altogether.

SpicyDice, 2021-03-12:
Method 2 solved my issue completely.

You have my thanks for providing the sources, gathering the information and putting them altogether.

Xayanide, 2021-03-12:
## December 16, 2018 - Default Tablet Configuration of G640s Unchanged in TabletDriverV0.2.3
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00
```

## January 12, 2021 - Found a Solution for Tablet Init failed

Sources, Credits, References: 

Issue Numbers 
#941 
#990 

Method 1
https://github.com/hawku/TabletDriver/issues/990#issuecomment-778399795
Method 2, Method 3 
https://github.com/hawku/TabletDriver/issues/941#issuecomment-758519839
https://github.com/hawku/TabletDriver/issues/990#issuecomment-778567638

## Method 1

Find ```XP-Pen G640s``` in ```TabletDriverV0.2.3\config\tablet.cfg```
Change InitReport to 
```InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00```
Save changes and restart driver.

Method 1 Config:
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

## Method 2
If that method did not help you, in my case, it didn't, here is another method that did solved my issue, hopefully it works for you as well.

Change ReportLength to
```ReportLength 12```

Replace InitReport with 
```InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00```

Method 2 Config:
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

## Method 3
However, if it still didn't work for you and still gave you the error, try these below.

```ReportLength 10``` 
```InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x0```

Method 3 Config:
```
#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 10
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x0
```

## Post Scriptum

You can copy paste the Config code blocks and paste it into your tablet.cfg file, find G640s section and paste it there.

Alternatively just change each ```ReportLength``` and ```InitReport``` since those two are the only ones needed changes.

I made a ZIP file, inside are folders with ```tablet.cfg``` files.
Retaining every configuration of tablets,
But only changed the G640s section.

Attachment:
[G640s.Tablet.Configs.zip](https://github.com/hawku/TabletDriver/files/6132257/G640s.Tablet.Configs.zip)

It has folders namely ```Method 1```, ```Method 2```, ```Method 3``` with ```tablet.cfg``` files. 
With different configurations being made according to the methods above, this is just an alternative if you are lazy copy pasting texts and just want to copy paste and replace the config folder and the files inside altogether.

Steps:
1 Open ```G640s.Tablet.Configs.zip``` or extract it & move to Step 3
2 Choose and Open any of the folders namely ```Method 1```, ```Method 2```, ```Method 3```
3 Copy or Move the ```config``` folder.
4 Paste it to your ```TabletDriverV0.2.3``` folder.
5 If it asks you to Replace the folder or ```tablet.cfg```
6 Replace it.
7 Run ```TabletDriverGUI.exe```
If the Error appears, try the other Method folders and do the steps again.

SpicyDice, 2021-03-12:
Method 2 solved my issue completely.

You have my thanks for crediting the sources, gathering all the information and putting them altogether.

closed 2021-03-12 by SpicyDice