hawku/TabletDriver #941: XP-Pen G640S Tablet Init Failed

Trizen-Git, 2020-12-28:
So every time I try to start this program with my tablet it recognizes the tablet, but is unable to start.

[Error Log Here](https://pastebin.com/Su51iegT)

I've installed the VMulti drivers (tested both with the XP-Pen drivers and with TabletDriverGUI drivers)
I've done the thing where you have to end the Pentablet.exe and PentabletService.exe tasks as well as delete the files, and I cannot seem to get it working.

I'll use XP-Pen's drivers for now, but please if you could get back to me as soon as possible that would be great, thanks!

## Hardware/software information
Windows Version: Microsoft Windows NT 6.2.9200.0

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G640S

Neoalter, 2020-12-31:
Same here, i;m getting the same problem, i hope the devs or someone can provide a fix for the issue

Trizen-Git, 2021-01-12:
Update: Got it working, changed to this in the tablet.cfg
`#
# XP-Pen G640s
#
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640s"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00`

Trizen-Git, 2021-01-12:
Found a solution:

- Go to the config folder where you installed the TabletDriver
- Open tablet.cfg in a file editor such as Notepad++
- Find XP-Pen G640s
- Replace code with:
```
HIDTablet 0x28BD 0x0906 0xFF0A 0x0001
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 32000
MaxY 20000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

- Save changes
- Restart drivers
- It works!

closed 2021-01-12 by Trizen-Git