hawku/TabletDriver #942: XP-Pen G640 tablet "Tablet not found"

miazaky, 2020-12-28:
So, I've done everything the tutorial "Installation" said in the website and the driver cant find my tablet. Tried several times but it didn't change anything. 

## Hardware/software information
Windows Version: Windows 10 Pro 2004

Tablet Driver version:  0.2.3

Tablet model: XP-Pen G640

Lenaros, 2020-12-28:
It's just that xppen once released another (but worse) version of G640, which I have too, and you probably do too, just replace this tablet.cfg settings (just delete it all, open it in notepad or something), and paste this in there:
HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
Should work after that

Nov4k, 2020-12-29:
I've done what you did but it still doesn't work

miazaky, 2020-12-29:
@Lenaros thanks friend it worked <3

closed 2020-12-29 by miazaky